GeekMagicClock/HelloCubic-Lite #10: cryptocurrency rate

alkawman, 2024-08-15:
How can I display information about BTC, LTC, etc? I just bought this thing today and didn't know there were different versions.

urtho, 2024-12-23:
IMHO this software is based on an open source GPL licence - so we can just request the source code and they cannot refuse. 
https://github.com/peng-zhihui/HoloCubic/blob/main/LICENSE

GeekMagicClock, 2024-12-25:
Hello, buy before you read the specs, make sure no misleading.

closed 2024-12-25 by GeekMagicClock

Eth030, 2024-12-25:
> IMHO this software is based on an open source GPL licence - so we can just request the source code and they cannot refuse. 
> 
> https://github.com/peng-zhihui/HoloCubic/blob/main/LICENSE

How can we proceed with that, I would love to add an crypto scene for the halo lite.