GeekMagicClock/HelloCubic-Lite #12: Can’t use gif

Wop265, 2024-09-12:
For some reason it keeps on showing the jellyfish and I’m trying to upload my own gif I put in the exact dimensions still not working, help!

GeekMagicClock, 2024-09-13:
@Wop265  Hi, access the web console, and show us the file list under the picture options please.

Wop265, 2024-09-13:
> @Wop265  Hi, access the web console, and show us the file list under the picture options please.

I’m trying to view the web console but this is all it shows, it won’t even load and there’s no WiFi network to click. And it’s still showing a jelly fish even when I unplug and plug back in 
[image][image]

GeekMagicClock, 2024-09-18:
Hi, you can try this, plug it on, and search for hotspot "GIFTV", and connect to it, and re-configure your settings, change the photo etc, the "GIFTV" will be off if there's no connection in 1 minute.