GeekMagicClock/HelloCubic-Lite #15: Cube Glass broken. Flip Screen

mjimeneznet, 2024-09-28:
My cube glass was broken by my cat :(

Could you add the option in settings to flip the screen and use it as a mini screen? I would like to use it as mini screen

Thanks

joeyparis, 2024-10-03:
It would also be nice for viewing from the top of the cube instead of the front!

ifengchao, 2024-10-04:
@mjimeneznet @joeyparis 
http://your_cube's_ip/set?rot=2
Hello, this is not an officially supported method, but can be used to view the screen from the top.

mjimeneznet, 2024-10-04:
> @mjimeneznet @joeyparis http://your_cube's_ip/set?rot=2 Hello, this is not an officially supported method, but can be used to view the screen from the top.

That worked for me. Is there any way to know what endpoints this device has?

joeyparis, 2024-10-04:
@mjimeneznet I'm working on a list now based on the network requests made from the web interface. That being said, some endpoint like ?rot= don't appear so I'm sure there's still many I'm missing