GeekMagicClock/HelloCubic-Lite #16: How to connect to home wifi network

Larsvb0, 2024-09-28:
If I hit scan I can see my 2.4G network. I add the password, the device reboots and then uses its own Wifi GIFTV again. I have WPA/WPA2_personal. It just does not connect. Any ideas?

GianBRSP, 2024-12-02:
I have similar problem here: I can connect but no more events appears. Only the initial message in the cube... Where I can have techncal suppport?

GeekMagicClock, 2024-12-02:
Hi, there're steps in the user manual, have you followed the FAQ part, to narrow down the problem?

derMiepz, 2024-12-04:
I have the same problem? Could it a problem with the SSID having a space in its name like "My WIFI"?

derMiepz, 2024-12-05:
Yeah I tried it later that evening with an wifi extender I had laying around. It was the space. I created a new wifi without a space in the name and it worked first try.

GeekMagicClock, 2024-12-06:
> Yeah I tried it later that evening with an wifi extender I had laying around. It was the space. I created a new wifi without a space in the name and it worked first try.

Thanks for the report, this need to be fixed if it is confirmed.

GeekMagicClock, 2024-12-06:
> I even tries with named Test on 2.4ghz, channel 1, no encryption and still did not work. I am using Synology mesh routers so that might be the case.

Mesh router settings maybe complicated, but should work, what is the wireless mode, is it 802.11 b/g/n ?

Jackson3006, 2025-06-25:
I have same problem after install a Mesh system with two AVM 7590. Without Mesh connection working. All other IOT devices will work with the Mesh system!