GeekMagicClock/HelloCubic-Lite #19: [WIP] API Endpoints

joeyparis, 2024-10-04:
# WIP - API Endpoints

It would be awesome if we could get an official list of API endpoints for reading settings as well as setting them. I've been able to reverse-engineer some of them via the network logs on the interface, but an official list would be great!

This is a work in progress. I believe I've gotten all of the acces

Here is what I have compiled so far:

--

## WiFi

### Scan WiFi
#### `GET /wifi.json`

#### Example Request

```bash
CURL -X GET http://192.168.X.X/wifi.json
```

#### Example Response

```json
{
  "aps": [
    {
      "c": "1",
      "ss": "WiFi Network 1",
      "e": 1,
      "r": -79
    },
    {
      "c": "6",
      "ss": "WiFi Network 2",
      "e": 1,
      "r": -73
    },
    {
      "c": "6",
      "ss": "WiFi Network 3",
      "e": 1,
      "r": -79
    },
    {
      "c": "6",
      "ss": "WiFi Network 4",
      "e": 1,
      "r": -85
    }
  ]
}
```

### Save WiFi credentials
#### `GET /wifisave`

#### Parameters

| Name | Description | Type |
| --- | ---- | ---- |
| s | SSID | string |
| p | Password | string |

#### Example Request

```bash
curl -X GET http://192.168.X.X/wifisave?s=NETWORK_SSID&p=NETWORK_PASSWORD
```

### Delay WiFi Connection
#### `GET /set`

How long does it take to start connecting to the WiFi after starting up:
It's useful when used in car and the WiFi of the car starts slowly, or used at home when the router is powered off and restarted at regular intervals.

#### Parameters

| Name | Description | Type |
| --- | ---- | ---- |
| delay | Wifi connection delay in seconds | integer |

#### Example Request

```bash
curl -X GET http://192.168.X.X/set?delay=10
```

## Weather

### Set weather city

Location example, the capital city of Korea, please input: Seoul

[optional, but if there are several same city names in your country/region, you need to find the correct city code.]
**Tips：Where to find Your Own city code >>>**

[1.Click me to search your city code after internet is connected](https://openweathermap.org/find)

2.Input your city name, click search and then choose the result, open it, the number at the link address tail is your city code! eg: Seoul,KR https://openweathermap.org/city/1835848, 1835848 is city code of Seoul,KR.

#### Parameters

| Name | Description | Type |
| --- | ---- | ---- |
| cd1 | City code from [OpenWeatherMap.com](https://openweathermap.com) | string |
| cd2 | Unsure? Web client always sets to 1000. Could be secondary city | string |

#### Example Request

```bash
curl -X GET http://192.168.X.X/set?cd1=1835848&cd2=1000
```

### Set weather units

#### Parameters

| Name | Description | Type | Accepted Values |
| --- | ---- | ---- | ---- |
| w_u | Wind Speed Unit | string | `m/s`, `km/h`, `mile/h` |
| t_u | Temperature Unit | string | `%C2%B0F`, `%C2%B0C` |

#### Example Request

```bash
curl -X GET http://192.168.X.X/set?cd1=1835848&cd2=1000
```

---

## Scratch Pad
Everything below here is my scratchpad of observed endpoints. I'm just putting it here temporarily so as I properly format everything others can reference it to start working on their own projects.

```

# Wifi Credentials
/wifisave?s=SSID&p=PASS

# Wifi Delay
/set?delay=0

# Set rotation
/set?rot=0
1
2
3
4 # default

# Set weather city, cd2 is always 1000?
/set?cd1=CITY_CODE&cd2=1000

# Set weather units
/set?w_u=m/s&t_u=%C2%B0C&p_u=hPa

w_u:
 - m/s
 - km/h
 - mile/h

t_u:
 - %C2%B0F (Fahrenheit)
- %C2%B0C (Celsius)

p_u:
 - hPa
 - kPa
 - mmHg
 - inHg

# Set weather update interval in minutes
# /set?w_i=20

# Set weather API key
/set?key=API_KEY

# Set Time number color
/set?hc=%23ffffff&mc=%23feba01&sc=%23ff5900

# Set 24 hours time format
/set?hour=0 # 24 hour
/set?hour=1 # 12 hour

# Set date format
/set?day=1 # DD/MM/YYYY
/set?day=2 # YYYY/MM/DD
/set?day=3 # MM/DD/YYYY
/set?day=4 # MM/DD
/set?day=5 # DD/MM

# Set colon blink
/set?colon=0 # No blink
/set?colon=1 # Blink

# Set time font
/set?font=1 # Default Big Font
/set?font=2 # Digital Font

# Set NTP server domain
/set?ntp=pool.ntp.org

# Set countdown timer
/set?yr=YYYY&mth=MM&day=DD

# Set daylight savings time
/set?dst=0 # Disable
/set?dst=1 # Enable

# Enable image auto display and interval
/set?i_i=5&autoplay=1

# Set theme
/set?theme=1 # Weather clock today
/set?theme=2 # Photo album
/set?theme=3 # Time style 1
/set?theme=4 # Time style 2
/set?theme=5 # Time style 3
/set?theme=6 # Days countdown
/set?theme=7 # Simple weather clock

# Set autotheme switch
/set?theme_list=0,0,0,0,0,0,0&sw_en=1&theme_interval=10 # Disable

# Set brightness
/set?brt=255

# Set nightmode
/set?t1=22&t2=6&b1=50&b2=73&en=1

# Reset to factory default
/set?reset=1 # Always returns a failed status even when successful

# Reboot
/set?reboot=1 # Always returns a failed status even when successful

# Get device storage
/spaces.json

# Set image
/set?img=%2Fimage%2F<IMAGE_NAME>

# Delete image
/delete?file=%2Fimage%2F<IMAGE_NAME>

# List images (as html)
/filelist?dir=/image

# Upload image
/doUpload?dir=/image
- Form data - file: binary

# Set weather clock gif
/set?gif=%2Fgif%2f<IMAGE_NAME>

# Upload gif
/doUpload?dir=/gif
- Form data - file: binary

# Status
/album.json
/app.json
/brt.json
/city.json
/colon.json
/config.json
/day.json
/daytimer.json
/delay.json
/dst.json
/font.json
/hour12.json
/key.json
/ntp.json
/space.json
/theme_list.json
/timebrt.json
/timecolor.json
/unit.json
/v.json
/w_i.json
/wifi.json

```

the3dsandwich, 2024-12-04:
Hi, thanks for the information!
I just got mine a couple days ago and stumbled across your thread, may I ask what was your method reverse engineering it?
Maybe I can help discover something too

Also, off topic but how did you verify that rotation works? my cube replied FAIL each time I call via `curl -X GET 'http://10.1.0.113/set?rot=2'`

Lastly, this is a ESP32-WROOM-32E right? Did you figure out how to update firmware if needed?

MetalRain682237, 2025-01-15:
Thank you for this @joeyparis!
Any idea how to write custom text to the screen?
Otherwise my idea was to create jpeg images with the text on them and set it as the current image.

dkgoodman, 2025-04-30:
I also wish I could write custom text to the screen. I'm in the middle of writing a python script that does almost everything the web page does, and I plan to add a function that will let me write text on an image or colored background and send it to the device. This way it could display things programmatically.

MetalRain682237, 2025-07-10:
> I also wish I could write custom text to the screen. I'm in the middle of writing a python script that does almost everything the web page does, and I plan to add a function that will let me write [...]

Yeah that's a good idea, I used JavaScript but was able to write custom text into a gif or png and upload it