GeekMagicClock/HelloCubic-Lite #21: cant upload gif

IocknIoad, 2024-11-13:
I'm trying to upload a custom gif, when i press the upload button the page loads and nothing happens
i turned translate off and resized the gif to 240 but I'm still having the same issue

GeekMagicClock, 2024-11-16:
1. Is the free space enough?
2. Did you use translation on the web? Better to disable it.

phanindrabv, 2024-12-07:
Same happening with me, disabled translation on web, But still can't upload GIFs. It just says "Saved successfully" but wouldn't show on the screen.

GeekMagicClock, 2024-12-08:
Can you show me your file list? and what is on the screen?

CKCMorgoth, 2024-12-11:
Okay so I am a bit confused with this "translate on the web" part
Is this a setting in the cube I cannot find?
Or is this like the chrome intern translation feature?
I tried uploading gifs from firefox and chrome, I have no automatic translation on, but I cant upload gifs
Can you please give more information or a step-by-step?
or maybe is it possible to upload by ftp?
thanks

GeekMagicClock, 2024-12-12:
The first method would be to downsize the file size, you can use the optimize/cut options here on ezgif.com.
And the second, you can delete the gif files under the weather option.

Not possible to extend the size without re-create the hardware.

GeekMagicClock, 2024-12-12:
> I had problems uploading but I figured out the files were simply too big is there an option to increase the storage size maybe? or will there be a bigger version?

The first method would be to downsize the file size, you can use the optimize/cut options here on ezgif.com.
And the second, you can delete the gif files under the weather option.

Not possible to extend the size without re-create the hardware.

GaryHeat, 2024-12-31:
Having same issue with 404 error.  Steps followed so far:

Browser: Chrome
Translation: OFF
downsized image to 240x240
cleared all current images
downgraded firmware to 7.0.16 (from 7.0.17)

All other functionality is working. Please advise

GeekMagicClock, 2025-01-06:
> yeah but cut gifs wont be shown fullscreen, I guess there is no "fullscreen" option? When optimizing, I get strange green artefact errors on the gifs
> 
> I will try the weather gif delete option
> 
> Would it be possible to kinda stream gifs on it, use it as a screencast device? Or upload new gifs and delete old ones in a loop? Just some ideas :)

If the gif size is 240x240px, then it is fullscreen, you could delete one then upload a new one, but I wonder in what kind of situation you need this operation.

GeekMagicClock, 2025-01-06:
> Having same issue with 404 error. Steps followed so far:
> 
> Browser: Chrome Translation: OFF downsized image to 240x240 cleared all current images downgraded firmware to 7.0.16 (from 7.0.17)
> 
> All other functionality is working. Please advise
Hello,
Is this always happens?  Have you tried to clear the web broswer cache?

GaryHeat, 2025-01-06:
> Having same issue with 404 error. Steps followed so far:
> Browser: Chrome Translation: OFF downsized image to 240x240 cleared all current images downgraded firmware to 7.0.16 (from 7.0.17)
> All other [...]

I did the following steps this morning and it seems to have resolved.

1. factory reset of device
2.  unplugged from power for 2 days (not purposely this long, just happened to be 2 days as I was doing other things)
3. cleared browser cache as recommended

It is now allowing for pictures and gif's to upload

Thank you for support