GeekMagicClock/HelloCubic-Lite #23: offline mode doesnt supported

lvlichii, 2024-11-22:
Hi, I onboarded the device today in my network but it constantly reboots. Per default all my "IOT" devices doesnt have access to the internet and I block everything. In the packet capture its clear to see that the device constantly tries to contact openweather to get the weather data. If I block this traffic the device reboots after a few secs. If I set a allow rule on my firewall the device stays alive. Can you please add a offline option or check why it crashes if its not wished?

lvlichii, 2024-11-22:
after a deeper look it seems you guys do something strange with the software^^

Why does my cube talk to https://www.malwareurl.com/listing.php?domain=146.185.152.20?

146.185.152.20:
yginy.hair

this is more than suspicious

GeekMagicClock, 2024-11-23:
The photo album mode supports offline mode, no internet needed, you can set it to this mode.
But if you need weather or time display correctly, the internet is needed.

As we set weather service as a backgroud base service, will check the weather at certain interval, but if the internet is not available, the cube should stop trying but not reboot, that maybe a bug.

If you are concern about safety, you can use the photo album mode only, and no need to configure the WiFi info.

As for the link you mentioned "https://www.malwareurl.com/listing.php?domain=146.185.152.20" before, please confirm that it is sent by cube and share a complete evidence chain, because we have not designed this function, but we cannot rule out that some third-party libraries have requested this link.

lvlichii, 2024-11-23:
[image]

lvlichii, 2024-11-23:
I used the firmware from this git

lvlichii, 2024-11-23:
[image]

another capture just a few mins ago. It seems the device tries only to connect every ~10 mins

So I cant see constantly traffic towards to the location but it defenitly tries to open a session!

lvlichii, 2024-11-23:
I did not allow the traffic so I cant see what he tries to send. Its port 80 as destination so maybe a http put or post to send userdata oder wifi passwords towards a fishi location. Maybe I will build later a special SSID and vlan to capture what the devices tries to share when I allow the traffic towards the location^^ If its really not you guys who implemented this then you have probably a security incident

k0nradk, 2024-11-26:
As far as I understand, it doesn't seem to by anything malicious. The address leads to https://openweathermap.org/api. It's the source for the weather info.

lvlichii, 2024-11-26:
Trying 146.185.152.20:80...
* Connected to 146.185.152.20 (146.185.152.20) port 80
> GET / HTTP/1.1
> Host: 146.185.152.20
> User-Agent: curl/8.9.1
> Accept: */*
>
* Request completely sent off
< HTTP/1.1 301 Moved Permanently
< Server: openresty
< Date: Tue, 26 Nov 2024 17:35:52 GMT
< Content-Type: text/html
< Content-Length: 166
< Connection: keep-alive
< Location: http://openweathermap.org/api
<
<html>
<head><title>301 Moved Permanently</title></head>
<body>
<center><h1>301 Moved Permanently</h1></center>
<hr><center>openresty</center>
</body>
</html>
* Connection #0 to host 146.185.152.20 left intact

Its a redirect^^ in case I would do suspicious stuff with my firmware I would also simply do then a redirect from the suspicious address to openweather to say its only openweather^^

Other question: Do you think openweather register this domain?

146.185.152.20 - yginy.hair 

If you have any official infos that this IP address is used by openweather please let me know. When I search the IP address I get only results like above from malwareurl.

Makes no sense that openweather buys a domain yginy.hair and use it only for redirect to the official page.

I hope im wrong but for my highly suspicous^^

lvlichii, 2024-11-26:
okay im wrong:

[image]

there is a response from DNS with this IP. 

So dont ask me why this domain is regsitered:
146.185.152.20 - yginy.hair

But it seems to be valid. Sorry for this revolt but for such topics its better to be paranoid

GeekMagicClock, 2024-11-27:
I've been too busy recently, but I see it has been resolved. Thank you for all your clarification!