GeekMagicClock/HelloCubic-Lite #25: Switching languages and region

Tony1881ynoT, 2024-12-10:
Hi Team,

I've got my cubic from China and the interface is all in Chinese without the ability to select international cities...

Is there a way to update the firmware to have it switch to the "International" version please?

Thanks

GeekMagicClock, 2024-12-10:
Hello, what it is displayed?

Tony1881ynoT, 2024-12-10:
It’s all in Chinese so when I put in Auckland as the city and then I refresh the page, it just reverts back into the original default.
Current version is as below, it won’t update to the English version.

Thank You and Kind Regards,TonyW

GeekMagicClock, 2024-12-11:
@Tony1881ynoT  Hello, that is different model, https://github.com/GeekMagicClock/HelloCubic-V2 here's the english version, but I don't know if this can be updated to chinese version, since we do not sell chinese version directly, you may need to ask the retailer.

closed 2024-12-11 by GeekMagicClock