GeekMagicClock/HelloCubic-Lite #26: Webp support

MaxSteven, 2024-12-10:
Hi I just got one today.
It's quite cool but can you update so it can support webp gif?
The memory is so low so traditional gif is really limited.
webp can provide smaller file size with better quality :(

GeekMagicClock, 2024-12-11:
Yes, I have the same idea before, but don't know how to decode and display webp format on the device yet.

GeekMagicClock, 2024-12-12:
https://community.platformio.org/t/undefined-reference-errors-when-using-libwebp-library/21026
Sorry no opened bin, but seems possible.

MaxSteven, 2024-12-13:
that's great I just want to help that's all
do you consider to create a product which has bigger size?

GeekMagicClock, 2024-12-15:
> that's great I just want to help that's all do you consider to create a product which has bigger size?

Yes, Cube-Max, 50x50x50mm size.
https://github.com/GeekMagicClock/Cube-Max

MaxSteven, 2024-12-19:
I found it. But since it's only 3Mb I hope the next version can allow use micro sd card for larger storage