GeekMagicClock/HelloCubic-Lite #27: Opening up the source code

paaland, 2024-12-11:
Can you please consider opening up the source code so you can receive help adding functionality and fixing bugs?
That would also close any issues regarding this device doing strange requets to chinese addresses which is why I have it isolated on a separat vlan with no access to my other devices. I simpy don't trust it.

If I knew this device was closed source I'd probably not have bought it.

GeekMagicClock, 2024-12-12:
Hi, we never claimed open source anywhere anytime, it's on your own, if you don't trust it, I would say nothing, all on your own.

But we claimed here to others, we didn't send any "personal info" to China or any other country, all http/tcp data  can be captured, no our own server inside until now, but possible for future products for better product experience, like update firmware on device.

paaland, 2024-12-12:
I just think this thing could be so much better if you got help from other developers via pull requests.

HmTici, 2024-12-22:
is there any way to make my own dashboard ?

urtho, 2024-12-23:
IMHO this software is based on an open source GPL licence - so we can just request the source code and they cannot refuse. https://github.com/peng-zhihui/HoloCubic/blob/main/LICENSE

IVANrn4hag, 2024-12-23:
Сделайте пожалуйста еще возможность запрашивать какие либо API что бы можно было засунуть туда что то свое, пробки или курс акций и т.д. Раз исходный код закрыт.

GeekMagicClock, 2024-12-25:
> IMHO this software is based on an open source GPL licence - so we can just request the source code and they cannot refuse. https://github.com/peng-zhihui/HoloCubic/blob/main/LICENSE

NO, actually you don't know about it, otherwise you would not have made this wrong judgment about this model, they are different, again this is not open source for now, but possible in future.

GeekMagicClock, 2024-12-25:
> Сделайте пожалуйста еще возможность запрашивать какие либо API что бы можно было засунуть туда что то свое, пробки или курс акций и т.д. Раз исходный код закрыт.

Sorry, no plans yet.

GeekMagicClock, 2024-12-25:
> I just think this thing could be so much better if you got help from other developers via pull requests.

Appreciate your thoughts, but I don't want to complicate things at this moment, It's beyond my current ability.

closed 2024-12-25 by GeekMagicClock