GeekMagicClock/HelloCubic-Lite #42: internet and Disconnect Issue

AmberDweeb, 2025-03-01:
I went to the webpages' Network section, erased my wifi on two columns and then the website bugged out and it says IP unset on the screen. I was trying to change the wifi, didn't think I had to always have a wifi connected to the cube. Is there a way to have the cube offline and is there anything I can try to fix my display?

https://youtube.com/shorts/5BbSS06s3h0?si=Dr5Gqz72ysK2hr_n This is what it looks like, just plays for one loop and turns off.

GeekMagicClock, 2025-03-02:
Hi, you could try a factory reset according to the manual.

AmberDweeb, 2025-03-02:
> Hi, you could try a factory reset according to the manual.

I've tried resetting according to the manual by plugging the power supply in and out once I see the progress bar many, many times as listed but it didn't work.

GeekMagicClock, 2025-03-03:
> > Hi, you could try a factory reset according to the manual.
> 
> I've tried resetting according to the manual by plugging the power supply in and out once I see the progress bar many, many times as listed but it didn't work.

Please unplug the power immediately when the progress bar just starts to go and repeat the operation twice. You will definitely be able to do it.

AmberDweeb, 2025-03-03:
Already tried that, doesn't change anything, I've lost count how many times I've tried doing it

AmberDweeb, 2025-03-14:
So I went to the website, went to the network page, where you input the wifi (I had one connected and decided to disconnect it) the device said "ip unset" when booting up. I couldn't factory reset based on the steps written on the GitHub, wifi still showed up from it but would disappear after a minute or so. I decided to change its up to my wifi's ip, then decided to connect to it on my Samsung tablet, saw that I had an option to manage network, clicked it, sent me to the website. From there I put in the wifi and password and now it's all like a bad dream...

GeekMagicClock, 2025-03-14:
I think you may not reset it actually, did you press the reset factory button ? Can you access the console and show me your settings ?

AmberDweeb, 2025-03-14:
1. Ive already said the I've fixed it
2. There's no actual factory reset button, at least not on the outside, in the user manual of the cubic lite literally says reset/reconfigure (need to plug in then unplug the power supply right as the progress bar pops up, do it twice, and the third time it shouldve worked)

closed 2025-03-14 by AmberDweeb